ElectronicCats/Beelan-LoRaWAN #173: What is What , RFM_pins?

OLIVIERStephane, 2023-07-31:
Hello,

Thank you for your work and to reading me.

I don't find the good number for define RFM_pins:
const sRFM_pins RFM_pins = {
    .CS = 20,
        .RST = 9,
        .DIO0 = 0,
        .DIO1 = 1,
        .DIO2 = 2,
        .DIO5 = 15,
        };

I have Lora32 Lilygo, for communicate  I use (and it's work):
#define SCK 5
    #define MISO 19
    #define MOSI 27
    #define SS 18
    #define RST 14
    #define DIO0 26

But I don't know what is DIO0 or DIO1 or DIO2 or DIO5.

Can you help me?
Thank and have a nice day.

Eric286, 2023-08-01:
Hello @OLIVIERStephane!

Seeing the information on the repository [LilyGo-LoRa-Series](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series) for the Lora32 Lilygo board, the DIO0, DIO1, DIO2, and DIO5 pins are connected to the ESP32's GPIO pins as follows:

    DIO0: Connected to GPIO pin 26
    DIO1: Connected to GPIO pin 33
    DIO2: Connected to GPIO pin 34
    DIO5: Connected to GPIO pin 36

Also here I leave the descriptions for the DIO.

[image]

[image]

Kind Regards!

Support Team!

OLIVIERStephane, 2023-08-02:
Thank you for your answer.

```
    const lmic_pinmap lmic_pins = {
        .nss = 18, 
            .rxtx = 5,
            .rst = 23,
            .dio = {26, 33, 34} 
        };
```

Unfortunately, i stay with error:"RFM95 not detected"

Perhaps, there is not RFM95 on this card (I'm a beginner in Lorawan)

Eric286, 2023-08-02:
Hello @OLIVIERStephane!

This type of card does have an RFM95 module. Let's see what could be happening. First, could you tell me what specific model you have, since there are several models for Lora32 Lilygo.
If you don't know, send me a picture of it.

Also please see the official documentation for the board for more information, as there is more information on this there.
You can see the following repository.

https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series

Kind Regards!

Support Team!

OLIVIERStephane, 2023-08-03:
Thank you Eric for your help.

I just found a program that works.
I should get away with this.
thanks again

closed 2023-08-03 by OLIVIERStephane