ElectronicCats/Beelan-LoRaWAN #174: how to downlink class c?

halukmy, 2023-10-15:
the device is not listening sorry

YoshiDev22, 2023-10-16:
Hello, @halukmy!

Thank you for reaching out to us.

In order to understand better the problem that are you facing, please share with me the following information:
1. When sending a package, make sure that you have configured the confirmation of receipt of the package, for that the third parameter must be 1: `lora.sendUplink(myStr, strlen(myStr), 1, 1);`
2. Did you confirm that downlink messages are being sent every time the server receives data from your node?
3. Are the uplink messages stable and continuously reaching the server?
4. Which Activation method are you using (OTA or ABP)?
5. What band are you using?
6. What server are you using?
7. What frequency are you using?

Have a nice day!

YoshiDev22, 2023-10-20:
Dear @halukmy    

We will close this issue now, however, please feel free to reach out to us again in case you have further suggestions, encounter an issue, or you would like to contribute.

Wishing you a fantastic day ahead!
Sincerely,

closed 2023-10-20 by YoshiDev22