ElectronicCats/Beelan-LoRaWAN #179: DevNonce issue

piepak, 2023-12-20:
Hi there,

I'm using ESP32 boards + a LoRa module. Sometimes when my device wake up and reach the Network Server using the OTAA method, my console throw that type of error "devnonce has already been used". I think there might be two ways this could be solved. 

1. Modifying the `static void Generate_DevNonce(unsigned char *DevNonce)` in LoRaMAC.ccp file and add a few lines checking wether the code is compiled for an ESP32 to use it's embedded real number generator instead of the basic random() function from arduino library.

2. Add to the `LoRaWANClass` class a method that would allow us to setup up DevNonce for each joinRequest.

I hope my contribution will help improving this library.

Cheers.

closed 2024-02-07 by Eric286

areytechai, 2024-09-23:
any way to fixx?

sabas1080, 2025-03-25:
#210