ElectronicCats/Beelan-LoRaWAN #183: How to send an keep time as short as possible?

chromoxdor, 2024-03-22:
Hi,

i am building some LoRa Nodes based on Atmega328p + SX1276.
Since they are battery powered i tried to keep the on time as short as possible.

Joining works fine and afterwards sensor data is gathered and the device would go to sleep after a successful sending....

Here is a very simplified version of my code. The issue is of course the while loop.
How would i need to change the code to make this actually work? 
Thanks in advance!

```cpp
void setup() {
    // Set LoRaWAN Class change CLASS_A or CLASS_C
  lora.setDeviceClass(CLASS_A);
  // Set Data Rate
  lora.setDataRate(SF9BW125);
  // set channel to random
  lora.setChannel(MULTI);
  // Put OTAA Key and DevAddress here
  lora.setDevEUI(devEui);
  lora.setAppEUI(appEui);
  lora.setAppKey(appKey);
  tryJoining();
}

void gatherDataAndSend() {
    .
    .
    //get sensordata sensor 1
    .
    .
    //get sensordata sensor 2
    .
    .
    //send data
  String payload = makePayload();
  const char *payloadCStr = payload.c_str();
  size_t payloadLength = strlen(payloadCStr);

  Serial.print("Sending: ");
  Serial.println(payloadCStr);
  //sending
  lora.sendUplink(payloadCStr, payloadLength, 0, 1);

  // Check Lora RX
  bool isAck = false;
  sensorTime = millis();
  do {
    lora.update();
    isAck = lora.readAck();
  } while (!isAck || millis() - sensorTime < 2000);
  if (isAck) blinker(3);

  blinker(3);
  sleep()
}
```

Eric286, 2024-03-22:
Hello @chromoxdor!

Could you give me more information about what error happens in the while loop, or if it is just that the while loop in your gatherDataAndSend function keeps the device "awake"?

Have a nice day!

chromoxdor, 2024-03-22:
Hey @Eric286,

it is hard to describe since i am no expert in this. 
The loop basically wreaks havoc in the code....  
e.g. i have another while loop for a sensor before. The fist time it runs fine the second time after the "lora.update();" loop and the device went to sleep and woke up again to run the "gatherDataAndSend()" function,  the "sensor" looop runs forever ignoring its statement (also a timer). 
Sometimes the mcu reboots.
And apart from that no data is send.

Is rRunning "lora.update();" in a loop together with "lora.readAck();" even the right way to do since it seems to mess with the rest of the code?

(The whole code can be found here: https://github.com/chromoxdor/LoRa-and-LoRaWAN/blob/main/lora_2v1/lora_2v1.ino)

> Have a nice day!

You too... but here it is more of an evening :)

Eric286, 2024-03-25:
Hello @chromoxdor!

Since the loop continuously checks for an acknowledgment, the code execution stalls at this point, preventing other code sections (like the sensor loop) from running.

Is lora.update() and lora.readAck() Necessary in a Loop?

Yes, these functions are crucial for checking for incoming data and acknowledgments. However, using them within an infinite loop is problematic.

Implement serial communication to print debug messages during code execution to identify potential problems.

Please try to call lora.update() within the loop to check for available data.

Best Regards!

chromoxdor, 2024-03-27:
Thanks for your help. I adjusted my code and as soon as i come in reach of a gateway i´ll report back. 
Have a nice day!

Eric286, 2024-04-10:
Hello @chromoxdor 

How are you doing with your code?
Could you use a gateway?
Is there anything else I can help you with?

Best Regards!

chromoxdor, 2024-04-15:
Hi eric,

> How are you doing with your code?

Everything works fine now. After a lot of trial and error i realized, that there was a memory issues leading to unexpected behavior. 
The main steps i did that lead to success where:
1. reducing the `MAX_UPLINK_PAYLOAD_SIZE` in the config.h to 50
2. disabling serial communication completely

I also stripped the library a bit to just send data without waiting for an acknowledgement or any received data which saved space but more importantly time where the mcu can sleep.

> Could you use a gateway?

I bought one. :) I did myself and the community a favor...

Thanks and regards,
chromoxdor

Eric286, 2024-04-24:
Hello @chromoxdor 

I'm glad to know that everything worked.

If you encounter any further difficulties or wish to share any feedback, please feel free to reach out to us again. We are here to assist you.

Have a nice day!

closed 2024-04-24 by Eric286