ElectronicCats/Beelan-LoRaWAN #185: Correct LoRaWAN version (TTN)

miladinovic, 2024-04-08:
I am writing because there is some confusion regarding the correct LoRaWAN version to be selected. Upon researching, I found that the library is compliant with LoRaWAN Specification 1.0.3. However, in the code file **/arduino-rfm/lorawan-arduino-rfm.h**, it is defined as **#define LORAWAN_VERSION "1.0.0"**. So, I am wondering which version is correct.

If it is indeed LoRaWAN Specification 1.0.3 compliant, does the library somehow save the MAC frame count, or is the option for resetting MAC frame state needed to be selected in the Advanced MAC settings?

AndreaZGuz, 2024-04-08:
Hello @miladinovic 

This library is currently being updated. Most of the features are compatible with version 1.0.3. Some are already compatible with version 1.0.4 and are working properly. Testing and modifications will continue throughout the year.

If you are working with TTN, it is suggested to check the option "Resetting MAC frame state".

Is there any other question I can help you with?

miladinovic, 2024-04-08:
Thanks a lot! "Resetting MAC frame state" option in ABP works nicely, now the script works with TTN, although in a unidirectional manner, specifically from the node to the gateway. 

I tried to send bytes 88 88 88 88 88 bytes back to the node, and on the node serial I got _"Data received over RX1"_, but no string printed. I am using **class_a_abp.ino** example code.

Here is the screenshot from TTN. 
[image]

Any hint on why is that so?

AndreaZGuz, 2024-04-09:
Hi @miladinovic 

Class A opens two reception windows for the node, just after a message is sent.
Ensure that the payload you are sending from the server to the node matches any of these two windows. Otherwise, the message will be never received by the node.

AndreaZGuz, 2024-04-15:
Hi @miladinovic !

We will close this issue now, however, please feel free to reach out to us again in case you have further suggestions, encounter an issue, or you would like to contribute.
Wishing you a fantastic day ahead!

closed 2024-04-15 by AndreaZGuz