ElectronicCats/Beelan-LoRaWAN #196: DevNonce has already been used

IoTThinks, 2024-05-26:
The library takes a while with many Joinings to have JoinAccepted.
It needs to try its lucks with random devNonce in the Join Request.
[image]

By right, the library should save the devNonce persistently in the flash and increase devNonce when doing a new join.
[image]

areytechai, 2024-09-23:
i got also this problem, anyway to fix? or create random ?

areytechai, 2024-10-17:
Still waiting udpate, anyone?

Khaiminh2310, 2025-02-19:
Hi @IoTThinks, @areytechai 
Please see this comment [here](https://github.com/ElectronicCats/Beelan-LoRaWAN/issues/208#issuecomment-2667481889) to fix this issue on ESP32.

sabas1080, 2025-02-19:
Hello 

Thank you for reporting this.
This patch is available in the branch [Pullrequest 181](https://github.com/ElectronicCats/Beelan-LoRaWAN/pull/181).

If you are using ESP32, follow these steps to try the patch:

Uninstall the Beelan-LoRaWAN library from Arduino IDE. I suggest you also delete the "Beelan-LoRaWAN" folder from your Arduino > libraries directory.
Download the ZIP file from the branch [solve several problems](https://github.com/ElectronicCats/Beelan-LoRaWAN/tree/raul360-usbstick-ok). Go to Code > Download ZIP.
Open your Arduino sketch and go to Sketch > Include Library > Add .ZIP library and look for the .ZIP file you downloaded in the previous step.
Compile and upload the sketch.
Please let us know if you need further assistance.

sabas1080, 2025-03-25:
#210

closed 2025-03-25 by sabas1080