ElectronicCats/Beelan-LoRaWAN #197: Downlink no longer works

IoTThinks, 2024-05-26:
The device can join LoRaWAN network and I can send a downlink packet.
[image]

The device "hears" something but can not print it.
[image]

Last year, I tested downlink ok.
I suspect the new code broke the downlink.
https://github.com/ElectronicCats/Beelan-LoRaWAN/blob/master/src/arduino-rfm/lorawan-arduino-rfm.cpp#L506

Andermutu, 2024-06-12:
Hi @IoTThinks 

I am facing a similar issue, I am unable to receive downlink messages. In my case, I don't even get the "Data received over RX1" message. Is downlink broken? Did you find a solution?

bos2303, 2024-06-17:
I have two things conserning this same type of observation.
First, I use TheThingsNetwork TTN V3.

Hardware setup: Heltec WiFi Lora V2 hardware and bare otaa example from lib. Uplink functionality is flawless.
However downlink does not work unless two things are changed. First in  file LoRa-arduino-rfm.cpp  line 115 I have changed:
   LoRa_Settings.Rx1_Delay = 5000; // Thing stack seems to be 5000 ms (so Rx2_delay 6000 ms)
    LoRa_Settings.Rx2_Delay = 6000; // Rx2_Delay >= Rx1_Delay + RX1_Window
This to be compliant to TTN V3 ( tried to change ttn to 1000 mS but does not work and device will not join network with 1000mS in RX1 delay)
Second change:
In update() methode //Type A mote transmit receive cycle:
       if (Buffer_Rx.Counter != 0x00)
        {
            bool isConfirmed =  ((Message_Rx.MAC_Header & 0xE0)>>5) == 5 ? true : false ; // MType
            uint8_t fPort = Message_Rx.Frame_Port;
            if(lora.messageCallback) lora.messageCallback(&Buffer_Rx, isConfirmed, fPort);
            **//Buffer_Rx.Counter = 0x00; // clear counter for the next cycle
            Rx_Status = NEW_RX;**
            Serial.println("Data received over RX1");
            
        }
Bold is required  change, this is by no means the ’true fix’ but now it works for me. The only part I don’t understand is that the port is not having any effect or reflected in recv event. I would have expected som kind of port to be reflected in readData(outStr,fPort) or similar construction.

Andermutu, 2024-06-17:
> I have two things conserning this same type of observation. First, I use TheThingsNetwork TTN V3.
>
> Hardware setup: Heltec WiFi Lora V2 hardware and bare otaa example from lib. Uplink functionality [...]

Thank you, that fixes the issue. I have checked the release of July 2023, and the downlinks were working. `Buffer_Rx.Counter = 0x00;` was reset on the readData function and not on the update function.
I have also included some modifications to make easier decoding from cayenneLPP to JSON. I can confirm that with this modification the downlinks are working. I think this project is much easier to use and modify compared to LMIC, and should keep improving. Thanks.

[image]

[image]

REGIOIGER, 2024-06-25:
Thanks  [IoTThinks](https://github.com/IoTThinks) and [Andermutu](https://github.com/Andermutu). Your contributions are appreciated and we will continue to work to improve the library.

samuellbs, 2024-07-26:
It didn't work for me. I'm using ESP32 with library 2.5.0 and with the change in commit "fix esp32", and the code is the ABP example. The gateway sends the downlink and "Data received over RX1" appears.

reibuehl, 2024-09-09:
I just ran into the same issue with v2.5.0 which is the latest version available in the Arduino IDE library manager. I was able to fix it with the changes from the June 17 posts above but would like to know when there will be a release with the fix as this seems to be solved now since >2 months.

fedyfausto, 2025-01-07:
Downlinks still do not work! Is there a way to fix this? I read the post in this issue but I did not get the fix to be applied

sabas1080, 2025-02-19:
Hello 

Thank you for reporting this.
This patch is available in the branch [Pullrequest 181](https://github.com/ElectronicCats/Beelan-LoRaWAN/pull/181).

If you are using this library, follow these steps to try the patch:

Uninstall the Beelan-LoRaWAN library from Arduino IDE. I suggest you also delete the "Beelan-LoRaWAN" folder from your Arduino > libraries directory.
Download the ZIP file from the branch [solve several problems](https://github.com/ElectronicCats/Beelan-LoRaWAN/tree/raul360-usbstick-ok). Go to Code > Download ZIP.
Open your Arduino sketch and go to Sketch > Include Library > Add .ZIP library and look for the .ZIP file you downloaded in the previous step.
Compile and upload the sketch.
Please let us know if you need further assistance.