ElectronicCats/Beelan-LoRaWAN #202: DevNonce  alltime problem after try to joining platform

areytechai, 2024-09-29:
we have checked the DevNonce  number problem each time try to connect to platform,

kindly can you fix this problem?

Eric286, 2024-10-02:
Hello @mirhamza708 

Beelan-LoRaWAN relies on random DevNonce generation, which can lead to repeated values after resets, while the MCCI library ensures that DevNonce is unique and tracked properly.

Have a nice day!

areytechai, 2024-10-02:
> Hello @areytechai
> 
> Please try the following that one user mentioned to help with DevNonce.
> 
> ![99122915-3012e480-25cd-11eb-8bc3-2be08f81a7ee](https://private-user-images.githubusercontent.com/40640735/372919440-f3c50ac0-1762-4ce3-bbc0-889b3a7b6023.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc4ODczNDMsIm5iZiI6MTcyNzg4NzA0MywicGF0aCI6Ii80MDY0MDczNS8zNzI5MTk0NDAtZjNjNTBhYzAtMTc2Mi00Y2UzLWJiYzAtODg5YjNhN2I2MDIzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDAyVDE2MzcyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg5NWU0ZjM4ZTA5MDAyYjNiODA1Y2RlMGIzNGQ2ZDU5M2JkMjNlMzAxNDEwYWU4MTZlZGUxODVkZTQwM2JjNzQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.VtdfOg6aMl6JFNGlipzMASc1iZL2_EOglYgOWGN4les)
> 
> Have a nice day

Eric thank you so much,

can you share the whole sample code please, if it runs will so happy to find this fix!

areytechai, 2024-10-02:
[image]
[image]
[image]

unsigned char myDevNonce[2];

void myGenerateDevNonce(unsigned char *DevNonce) {
  unsigned int RandNumber = random(0x0000, 0xFFFF);
  DevNonce[0] = RandNumber & 0xFF;
  DevNonce[1] = (RandNumber >> 8) & 0xFF;
}

in setup()

  myGenerateDevNonce(myDevNonce);
 Serial.print("Generated DevNonce: ");
  Serial.print(myDevNonce[0], HEX);
  Serial.print(" ");
  Serial.println(myDevNonce[1], HEX);

but still

mirhamza708, 2024-10-04:
https://github.com/ElectronicCats/Beelan-LoRaWAN/issues/201

check this out my friend.

Eric286, 2024-10-10:
Dear @areytechai 

You can see this in the closed issue shared by mirhamza708.

Tell us if this worked for you or if you still have the same problem.

Have a nice day!

closed 2024-10-18 by xpeqex