ElectronicCats/Beelan-LoRaWAN #204: RFM95 not found

Finn-Land, 2024-10-15:
Hello,

I am sure this is a recurring question but I haven't been able to find an answer yet. I am currently working with the LilyGO TTGO ESP32 using the SX1276 chip. Right now I am trying to configure it as a class C device. I am just running the example code provided but I keep getting the error "RFM95 not found". I am using PlatformIO and I have changed the config file too.
Thanks in advance

`
// If your using PlatformIO you will need to set _CLASS_C_ using the -D flag in the build_flags options
#include <Arduino.h>
#include <lorawan.h>

const char *devEui = "0"; // replace with actual DevEUI
const char *appEui = "0"; // replace with actual AppEUI
const char *appKey = "0"; // replace with actual AppKey

const unsigned long interval = 15 * 1000; // 10 s interval to send message
unsigned long previousMillis = 0;         // will store last time message sent
unsigned int counter = 0;                 // message counter

bool sendack = false;
char myStr[50];

const sRFM_pins RFM_pins = {
    .CS = 18,   // j
    .RST = 14,  // j
    .DIO0 = 26, // j
    .DIO1 = 35,
    .DIO2 = 34,
};

void message(sBuffer *msg, bool isConfirmed, uint8_t fPort)
{

  char Buff[255];
  int size = msg->Counter;

  memset(Buff, 0x00, size + 1);
  memcpy(Buff, msg->Data, size);

  Serial.println("--------------------");
  Serial.print("Msg size as bytes : ");
  Serial.println(msg->Counter);
  Serial.print("Message :");
  Serial.println(Buff);
  Serial.print("Port :");
  Serial.println(fPort);

  if (isConfirmed)
  {

    Serial.println("ACK response Should be sent !");
    sendack = true;
  }
}

void setup()
{
  // Setup loraid access
  Serial.begin(115200);
  while (!Serial)
    ;
  if (!lora.init())
  {
    Serial.println("RFM95 not detected");
    delay(10000);
    return;
  }

  lora.setDeviceClass(CLASS_C);
  // Set Data Rate
  lora.setDataRate(SF8BW125);

  // set channel to random
  lora.setChannel(MULTI);
  // Put OTAA Key and DevAddress here
  lora.setDevEUI(devEui);
  lora.setAppEUI(appEui);
  lora.setAppKey(appKey);

  // Set Callback function
  lora.onMessage(message);

  // Join procedure
  bool isJoined;
  do
  {
    Serial.println("Joining...");
    isJoined = lora.join();

    // wait for 10s to try again
    delay(7000);
  } while (!isJoined);
  Serial.println("Joined to network");

  delay(1000);

  sprintf(myStr, "First Message");

  Serial.print("Sending: ");
  Serial.println(myStr);

  lora.sendUplink(myStr, strlen(myStr), 0, 1);
  counter++;
}

void loop()
{

  // Check interval overflow
  if (millis() - previousMillis > interval)
  {
    delay(1000);

    previousMillis = millis();

    sprintf(myStr, "Counter-%d", counter);

    Serial.print("Sending: ");
    Serial.println(myStr);

    lora.sendUplink(myStr, strlen(myStr), 1, 1);
    counter++;
  }

  if (sendack)
  {

    lora.sendACK();
    sendack = false;
  }

  lora.update();

  if (lora.readAck())
    Serial.println("ack received");
}
`

Finn-Land, 2024-10-15:
It is most likely a pin issue, but most examples I have found online use the ones I have used. I have tried several options but none seem to work.

AndreaZGuz, 2024-10-15:
Hello @Finn-Land,

Thank you for contacting us.
This issue is related to the pinout.
I suggest you review the official documentation for the board you are using, and, in addition, check the datasheet for the SX1276 LoRa transceiver, to confirm the pinout matching.

I am searching for information on the board you are using but I found different models. Do you have a picture from both sides of the board you can share?

Thank you.

Finn-Land, 2024-10-16:
Hello @AndreaZGuz ,

Thank you for getting back to me.  Below I have included the pictures of the ESP32. It comes with the attachable antenna by wire. I am using the pinout from the following website:

https://makeradvisor.com/esp32-sx1276-lora-ssd1306-oled/

It uses the same board as me but it does not work. Do you have any other ideas? The SX1276 datasheet does not tell me much since it should be the LilyGO TTGO ESP32 boards pinout that I should be looking at. But that pinout does not work for me.

[image]
[image]
[image]

Finn-Land, 2024-10-16:
I believe the board is the V1, this is the pin mapping from another well-known library:
https://github.com/espressif/arduino-esp32/blob/master/variants/ttgo-lora32-v1/pins_arduino.h

AndreaZGuz, 2024-10-16:
Hello @Finn-Land,

I have reviewed the schematics for V1 and according to the DIO pins connected, the pinout would be:
const sRFM_pins RFM_pins = {
.CS = LORA_CS, //18
.RST = LORA_RST, //14
.DIO0 = LORA_IRQ, //26
.DIO1 = 33,
.DIO2 = 32,
};

Could you please try this way?
Let me know your outcomes.

Finn-Land, 2024-10-17:
Hello, thank you for getting back to me.

I found the issue, the device I initialized the PlatformIO project with was the wrong device. After changing the device type to `ttgo-lora32-v1` it works. Hope this helps someone.

Thank you @AndreaZGuz for helping!

closed 2024-10-17 by Finn-Land