ElectronicCats/Beelan-LoRaWAN #207: Device hangs while in single receive window 1 (RX1) right after uplink

Khaiminh2310, 2025-02-13:
**Describe the bug**
I am configuring the device using OTAA and Class C. Uplink and downlink work fine. Every minute I send an uplink. The problem is that after an unknown amount of time the device hangs at the **RFM_Single_Receive()** function in the RX1 window. It could be after a few hours or a few days.

**To Reproduce**
- Configure the device using Class C and interrupts via pins DIO0, DIO1, DIO2. 
- Perform uplink 1 time/minute with confirmed mode to receive ACK in RX1 window

**Expected behavior**
Uplink & downlink are stable and skip RX1 or RX2 if timeout occurs (RX1_Window, RX2_Window) immediately after uplink transmission and switch to continuous receive mode.

Note: 
- I added some debug logs in the LoRaMAC.cpp and RFM95.cpp files and determined that the device hangs at line [721](https://github.com/ElectronicCats/Beelan-LoRaWAN/blob/v2.5.0/src/arduino-rfm/RFM95.cpp#L721) in the RFM95.cpp file.
- After disable the interrupts on pins DIO0,1,2 and switching to using the interrupts on the register, the device operates stably.
_const sRFM_pins RFM_pins = { .CS = *, .RST = *, .DIO0 = -1, .DIO1 = -1, .DIO2 = -1, .DIO5 = -1, };_ 

[image]

**Some information about my project:**
 - MCU: ESP32-C3
 - Lora chip: RFM95W
 - Dev on Platform IO
 - Lib version: v2.5.0

Marcelol52, 2025-02-13:
Hi @Khaiminh2310,

First of all, let me thank you for the very detailed description; it helps a lot when replicating the issue. If you are working with the library and it fails after an indefinite amount of time, the issue could be related to the radio itself rather than the library. Have you checked if the radio is operational when the code hangs?

Khaiminh2310, 2025-02-14:
Thanks @Marcelol52 for your feedback. How can I check if the radio is working or not when the code hangs?

Marcelol52, 2025-02-18:
Hello @Khaiminh2310 you could try reseting the radio via the RESET pin while the code is stuck and checking if that fixes the issue. 

Something else to keep in mind is that for Class C you might want to use RFM_Continuous_Receiv instead of RFM_Single_Receive.

Marcelol52, 2025-02-25:
@Khaiminh2310 

We will close this issue now, however, please feel free to reach out to us again in case you have further questions, suggestions, encounter an issue, or you would like to contribute.

closed 2025-02-25 by Marcelol52