ElectronicCats/Beelan-LoRaWAN #212: use STM32WLE5JC - rak3172

jpayalateva, 2025-04-03:
Hello.
I would like to be able to use the STM32WLE5JC chip that comes with the rak3172, Cortex M4, and SX126x modules.

The chip in question is used in many libraries, for example radiolib, STM32LoRaWAN, LoRaSpaceLib

The SX126x is connected internally; radiolib uses it this way.

//STM32WLE5JC - rak3172
static const uint32_t rfswitch_pins[] = {PB8, PC13, RADIOLIB_NC, RADIOLIB_NC, RADIOLIB_NC};
static const Module::RfSwitchMode_t rfswitch_table[] = {
 {STM32WLx::MODE_IDLE, {LOW, LOW}},
 {STM32WLx::MODE_RX, {HIGH, LOW}},
 {STM32WLx::MODE_TX_LP, {LOW, HIGH}},
 {STM32WLx::MODE_TX_HP, {LOW, HIGH}},
 END_OF_MODE_TABLE,
};

Do you know how I could use your library with STM32WLE5JC?

greetings.

closed 2025-04-04 by Marcelol52

jpayalateva, 2025-04-06:
351 / 5.000
Hi Marcelo.
Sorry for the delay in answering.

I actually want to use the STM32WLE5JC chip and be able to compile on Platformio. The RUI3 API doesn't support Platformio.

The class_c_otaa.ino example compiles perfectly for this chip.

[image]

But I don't know how to assign pins on the STM32WLE5JC, since this chip has the SX126x internally.

Do you know how I could define the pins?

wero1414, 2025-04-07:
Hello @jpayalateva the microcontroller will not have any trouble, but the radio will give you problems from what i've been able to find online, this SoC is incluiding an SX1262 based IP, that turns into a problem when using our library since we only support the SX1276 or SX1272, 

In other topic what you want to do is to use the SUBGHZSPI inside the SoC, you can use that interface directly you can read more about this [here](https://www.st.com/resource/en/reference_manual/rm0453-stm32wl5x-advanced-armbased-32bit-mcus-with-subghz-radio-solution-stmicroelectronics.pdf) just look for the SUBGHZSPI interface